serhmarch/ModbusTools #3: DSSU666 read value problem

folny, 2024-03-24:
Hello

I am trying this program to read values from the DSSU666 meter, but the values are not in the correct form. I have everything set up correctly and the format is set to float. I think the problem is in the format, you will probably have to add another float32 format.

DTSU666 manual https://www.deyeinverter.com/deyeinverter/2022/06/10/chintthreephase-instructionmanual-dt(s)su666-y0.464.1002v1.5190617.pdf

serhmarch, 2024-03-24:
Hello
I guess your issue is your meter has non default 16 bit register order in 32 bit float value. You can easily change it adding new dataView item (or editing existing dataView item) using menu: Data->New Item(s)->Advanced tab, than choose Register order to "Most Signified First". 
If this do not help you can do the same but chaging Byte Order to "Most Signified First" as well.
[image]

folny, 2024-03-25:
Thanks for the quick reply, the council helped and everything is fine now.

serhmarch, 2024-03-25:
But it's better to do the following. 
Since your device most likely has the same order of 16-bit registers for 32-bit values, it is better to set this order in the device settings. And the default settings of DataViewItem are the same as those of the device to which they refer.
[image]

closed 2024-09-10 by serhmarch