serhmarch/ModbusTools #4: How to compile

KLelong, 2024-05-04:
I need a modbus-simulator and I found this project.
Unfortunately, the included binaries for ubuntu don't work for me. I'm running kubuntu 22.04LTS and first I had to install libqt5serialport5. When I start the server I get an error : 
`symbol _ZdlPvm, version Qt_5 not defined`
I googled and I found that the application might be compiled with a different version of gcc.
So I decided to compile my own version. 
I installed  libqt5serialport5-dev.
I ran `qmake ModbusTools.pro`
Then I ran `make all`
But now I'm stuck with this error : 
```
...
/usr/lib/qt5/bin/qmake -o Makefile core.pro
Project ERROR: Unknown module(s) in QT: help
make[1]: *** [Makefile:476: Makefile] Fout 3
...

```
I suspect I have to install libqt5help5-dev but I can't find that package.
I tried removing that part from core.pro, but then the compiler complains about a missing help-function. So now I'm asking for help.
Any suggestions ? Google didn't help.
Thanks.

serhmarch, 2024-05-06:
Hello. Try next command:
`$ sudo apt-get install qtbase5-dev qttools5-dev libqt5serialport5-dev`
It might help

KLelong, 2024-05-06:
Thanks,
I had to install qttools5-dev, now I can successfully compile and run. The rest of your suggestions were already installed.
Now learning how to use the applications ;-) .

closed 2024-05-06 by KLelong