serhmarch/ModbusTools #9: Different exception code in version 0.3.2

maroc81, 2024-10-08:
I'm using version 0.3.2 and got an error when trying to read from a modbus TCP device.

```
08.10.2024 16:57:22.876 'Device': Tx: 00 05 00 00 00 06 01 04 00 00 00 01 
08.10.2024 16:57:22.876 'Device': Rx: 00 05 00 00 00 03 01 84 02 
08.10.2024 16:57:22.877 'Device': Returned Modbus-exception with code 212
```

The device table "Status" column just shows a red box.   If I use an old client I get the output: `'Device': TCP. Returned Modbus-exception with code '2'` and the status column shows `BadIllegalDataAddress`.

serhmarch, 2024-10-08:
Ok. The trouble was within `ModbusLib` sub-project. I have already fixed it and now I'm updating the latest release

serhmarch, 2024-10-08:
Done

maroc81, 2024-10-10:
Confirmed everything is working now as expected!  Thank you and thanks for this incredible software!

closed 2024-10-10 by maroc81