serhmarch/ModbusTools #11: Error in register addressing (Send Message)

crazy4volvo, 2024-10-12:
Hello, I'm Modbus newbie so in fact I'm not sure if I'm right or wrong ;)

I'm trying to experiment with  chinese XY SK-120 DC/DC converter registers with your very useful software.
I noticed that in "Send Message" window the function "WriteSingleRegister" sends data to register address +1.
For example:
R/w registers 400001 and 400002 set voltage and current. They are read correctly, but when I try to set the voltage (reg. 400001) , the data is written to 400002 despite setting 4x and 1 in Write Data fields.  This is valid for all other register adresses I tested (backlight, on/off etc).  

Many thanks!

serhmarch, 2024-10-12:
Yes, you're right. Wrong offset. Thank you for the report. I'll fix it in the next release

crazy4volvo, 2024-10-12:
Thank you Serhii, take care.

serhmarch, 2024-10-13:
Issue is fixed and new v0.3.3 release is available.

crazy4volvo, 2024-10-13:
I confirm, works as expected.

closed 2024-10-26 by serhmarch