serhmarch/ModbusTools #13: V0.3.4 | Serial port cannot open

MLaudren, 2024-10-18:
When using either server or client, with version 0.3.4,
I have an error: 'Device 1': Failed to open serial port. Error code: 2
[image]

 Tried with different adapter and port, and also running as admin, but could not get it to work

If I do the same with version 0.1.5, all is working great.
Likely the new Modbus library have issue ?, as explain here: https://github.com/serhmarch/ModbusTools/issues/10#issuecomment-2407214361

serhmarch, 2024-10-18:
Hello. `Error code: 2` is the value returned by the standard WinAPI function `GetLastError()` and means `ERROR_FILE_NOT_FOUND` according to the Windows documentation.

As I see in Project window you're using `RTU:COM34`. `COM34` doesn't look like a valid serial port number.  I just repeated your settings in server v0.3.4 app on Windows and I get completely the same output that you have. And this is understandable because I really do not have serial port with the name `COM34` in my list of serial ports. 

Maybe you selected `COM3` and then accidentally pressed `4` and thus got `COM34`?

MLaudren, 2024-10-21:
Hi Serhii,

I am using a lots of serial cables/adapter, and as Windows assign them individually, this one got the com port number 34,
See in screenshot from Device manager.

[image]

Same adapter, using Com34 is working fine with older version as I said.

If you think of tests I can do to narrow the area/cause, let me know, and I'll do.

serhmarch, 2024-10-21:
Its hard to say what is cause of the problem. This error can occur only as result of  `CreateFile` standard WinAPI function execution:
```cpp
            ...
            d->serialPort = CreateFileA(
                d->settings.portName.c_str(), // Port name
                GENERIC_READ | GENERIC_WRITE, // Read and write access
                0,                            // No sharing
                NULL,                         // No security attributes
                OPEN_EXISTING,                // Opens existing port
                0,                            // Disables overlapped I/O
                NULL                          // No template file
                );

            if (d->serialPortIsInvalid())
            {
                return d->setError(Status_BadSerialOpen, StringLiteral("Failed to open serial port. Error code: ") + toModbusString(GetLastError()));
            }
            ...
```
https://github.com/serhmarch/ModbusLib/blob/63b40ec78d7bc81d4fa02a45b27020ddf3585154/src/win/ModbusSerialPort_win.cpp#L77

It might have something to do with the input parameters of the function.
First, I need to expand the error message in the next release, and then you can take new screenshots to see what values are passed to the function.

serhmarch, 2024-11-05:
Hello. Please try new v0.3.5 version. Lets see what it would be.

MLaudren, 2024-11-07:
Hi @serhmarch ,
I've done some tests, but still not working.

Server still cannot open the port
[image]

Client same as server
[image]

I'll try to test more later.
I didn't saw any logs I can share.

MLaudren, 2024-11-12:
Hi @serhmarch ,

I'm running the application on Win 11 Pro (23H2)
As for the hardware, I have some FTDI chip, B&B Electronics USOPTL4, and other 3rd party cables.

ar-alekseev, 2024-12-13:
I have the same issue. It connects to COM1 but not to COM39. 
[image]

marchino2000, 2024-12-13:
Hello, same problem here using RS485 adapter and COM11, moved adapter to COM1 and it works.
Regards

serhmarch, 2024-12-13:
Thanks to everyone who helped me identify and find the problem. I think I have fixed it. The problem was that Windows handles COM port names with 2 digits differently (`COM10`, `COM11`, etc.), so I had to change the code a bit. 

I guess I'll make a new release tomorrow with this fix.

serhmarch, 2024-12-14:
New release [v0.3.9](https://github.com/serhmarch/ModbusTools/releases/tag/v0.3.9) with the fixed issue.

closed 2025-01-01 by serhmarch