serhmarch/ModbusTools #16: Failed to run build.sh on Rocky Linux 9.4

snowuyl, 2024-11-14:
git clone https://github.com/serhmarch/ModbusTools.git
cd ModbusTools
chmod +x ./build.sh
./build.sh 
CMake Error: Could not read presets from /home/snowuyl/samba/workspace_BASH/ModbusTools: File not found
CMake Error: Could not read presets from /home/snowuyl/samba/workspace_BASH/ModbusTools: File not found

serhmarch, 2024-11-14:
Hello. You can copy `CMakePresets.json` from `examples` folder into ModbusTools source code root folder and then exec `build.sh` once again. `CMakePresets.json` contains individual settings for build such as output bin folder etc so its not included in repository by default.

snowuyl, 2024-11-14:
Thanks for your reply!
cp examples/CMakePresets.json .
./build.sh 
Preset CMake variables:

  CMAKE_BUILD_TYPE="Release"
  CMAKE_CXX_COMPILER="/usr/bin/g++"
  CMAKE_C_COMPILER="/usr/bin/gcc"

-- The C compiler identification is GNU 11.4.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
MB: QT is enabled!
MB: Try to link QT library: Qt5::Core
MBTOOLS: Start configure library 'core'
CMake Error at /usr/lib64/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5Help" with any
  of the following names:

    Qt5HelpConfig.cmake
    qt5help-config.cmake

  Add the installation prefix of "Qt5Help" to CMAKE_PREFIX_PATH or set
  "Qt5Help_DIR" to a directory containing one of the above files.  If
  "Qt5Help" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  src/core/CMakeLists.txt:18 (find_package)

-- Configuring incomplete, errors occurred!
gmake: Makefile: No such file or directory
gmake: *** No rule to make target 'Makefile'.  Stop.

serhmarch, 2024-11-14:
Try to built step by step:
https://github.com/serhmarch/ModbusTools/tree/main/src#building-modbustools-from-source-on-linux

snowuyl, 2024-11-15:
Thanks for your reply! The following is building script on Rocky Linux 9.4.
sudo dnf update
sudo dnf groupinstall "Development Tools"
sudo dnf install qt5-qtbase qt5-qttools qt5-qtdeclarative
qmake --version
sudo dnf install git
git clone --recursive https://github.com/serhmarch/ModbusTools.git
mkdir -p bin/ModbusTools
cd bin/ModbusTools
qmake ../../ModbusTools/src/ModbusTools.pro -spec linux-g++
make

The following is building log on Rocky Linux 9.4.
Last metadata expiration check: 0:19:20 ago on Fri 15 Nov 2024 01:52:01 PM CST.
Dependencies resolved.
Nothing to do.
Complete!
Last metadata expiration check: 0:19:21 ago on Fri 15 Nov 2024 01:52:01 PM CST.
Dependencies resolved.
========================================================================================================================================
 Package                         Architecture                   Version                           Repository                       Size
========================================================================================================================================
Installing Groups:
 Development Tools                                                                                                                     

Transaction Summary
========================================================================================================================================

Is this ok [y/N]: y
Complete!
Last metadata expiration check: 0:19:23 ago on Fri 15 Nov 2024 01:52:01 PM CST.
Package qt5-qtbase-5.15.9-10.el9_4.x86_64 is already installed.
Package qt5-qttools-5.15.9-4.el9.x86_64 is already installed.
Package qt5-qtdeclarative-5.15.9-3.el9.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
QMake version 3.1
Using Qt version 5.15.9 in /usr/lib64
Last metadata expiration check: 0:19:23 ago on Fri 15 Nov 2024 01:52:01 PM CST.
Package git-2.43.5-1.el9_4.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
Cloning into 'ModbusTools'...
remote: Enumerating objects: 2288, done.
remote: Counting objects: 100% (72/72), done.
remote: Compressing objects: 100% (44/44), done.
remote: Total 2288 (delta 38), reused 32 (delta 28), pack-reused 2216 (from 1)
Receiving objects: 100% (2288/2288), 25.87 MiB | 6.67 MiB/s, done.
Resolving deltas: 100% (1582/1582), done.
Submodule 'src/modbus' (https://github.com/serhmarch/ModbusLib) registered for path 'src/modbus'
Cloning into '/home/andyueng/samba/workspace_C++/ModbusTools/src/modbus'...
remote: Enumerating objects: 976, done.        
remote: Counting objects: 100% (234/234), done.        
remote: Compressing objects: 100% (143/143), done.        
remote: Total 976 (delta 153), reused 148 (delta 89), pack-reused 742 (from 1)        
Receiving objects: 100% (976/976), 44.50 MiB | 6.63 MiB/s, done.
Resolving deltas: 100% (656/656), done.
Submodule path 'src/modbus': checked out '5abfb851944647197790ead0f25b24872bcbb50c'
Info: creating stash file /home/andyueng/samba/workspace_C++/bin/ModbusTools/.qmake.stash
cd modbus/src/ && ( test -e Makefile.modbus || /usr/lib64/qt5/bin/qmake -o Makefile.modbus /home/andyueng/samba/workspace_C++/ModbusTools/src/modbus/src/modbus.pro -spec linux-g++ ) && make -f Makefile.modbus 
make[1]: Entering directory '/home/andyueng/samba/workspace_C++/bin/ModbusTools/modbus/src'
g++ -c -pipe -O2 -D_REENTRANT -Wall -Wextra -fPIC -DMODBUS_EXPORTS -DQT_NO_DEBUG -DQT_CORE_LIB -I../../../../ModbusTools/src/modbus/src -I. -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o Modbus.o ../../../../ModbusTools/src/modbus/src/Modbus.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -Wextra -fPIC -DMODBUS_EXPORTS -DQT_NO_DEBUG -DQT_CORE_LIB -I../../../../ModbusTools/src/modbus/src -I. -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o cModbus.o ../../../../ModbusTools/src/modbus/src/cModbus.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -Wextra -fPIC -DMODBUS_EXPORTS -DQT_NO_DEBUG -DQT_CORE_LIB -I../../../../ModbusTools/src/modbus/src -I. -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o ModbusObject.o ../../../../ModbusTools/src/modbus/src/ModbusObject.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -Wextra -fPIC -DMODBUS_EXPORTS -DQT_NO_DEBUG -DQT_CORE_LIB -I../../../../ModbusTools/src/modbus/src -I. -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o ModbusPort.o ../../../../ModbusTools/src/modbus/src/ModbusPort.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -Wextra -fPIC -DMODBUS_EXPORTS -DQT_NO_DEBUG -DQT_CORE_LIB -I../../../../ModbusTools/src/modbus/src -I. -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o ModbusSerialPort.o ../../../../ModbusTools/src/modbus/src/ModbusSerialPort.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -Wextra -fPIC -DMODBUS_EXPORTS -DQT_NO_DEBUG -DQT_CORE_LIB -I../../../../ModbusTools/src/modbus/src -I. -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o ModbusRtuPort.o ../../../../ModbusTools/src/modbus/src/ModbusRtuPort.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -Wextra -fPIC -DMODBUS_EXPORTS -DQT_NO_DEBUG -DQT_CORE_LIB -I../../../../ModbusTools/src/modbus/src -I. -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o ModbusAscPort.o ../../../../ModbusTools/src/modbus/src/ModbusAscPort.cpp
[... 338 lines cut ...]
  149 |             this->sz = 0;
[... cut ...]
mv -f libmodbus.so ../bin/libmodbus.so
mv -f libmodbus.so.0 ../bin/libmodbus.so.0
mv -f libmodbus.so.0.3 ../bin/libmodbus.so.0.3
make[1]: Leaving directory '/home/andyueng/samba/workspace_C++/bin/ModbusTools/modbus/src'
cd core/ && ( test -e Makefile || /usr/lib64/qt5/bin/qmake -o Makefile /home/andyueng/samba/workspace_C++/ModbusTools/src/core/core.pro -spec linux-g++ ) && make -f Makefile 
Project ERROR: Unknown module(s) in QT: help
make: *** [Makefile:93: sub-core-make_first-ordered] Error 3

serhmarch, 2024-11-15:
Looks like Qt Help module is not installed. First check it:
```console
$ whereis libQt5Help.so
```
To install it try something like:
```console
$ sudo dnf install qt5-qthelp
```

snowuyl, 2024-11-15:
$ whereis libQt5Help.so
libQt5Help.so:
$ sudo dnf install qt5-qthelp
Last metadata expiration check: 0:01:22 ago on Fri 15 Nov 2024 03:13:28 PM CST.
No match for argument: qt5-qthelp
Error: Unable to find a match: qt5-qthelp

serhmarch, 2024-11-15:
Sorry. I do not know exact command to install it on your OS. Your goal is to install qt5 help module.
As I see from your logs it might be located as:
```console
$ whereis libQt5Help.so
libQt5Help.so: /usr/lib64/libQt5Core.so
```

serhmarch, 2024-11-15:
Try this:
```console
sudo dnf install qt5-qttools-devel
whereis libQt5Help.so
```

snowuyl, 2024-11-18:
Thanks for your suggestion.
sudo dnf install qt5-qttools-devel
whereis libQt5Help.so
libQt5Help.so: /usr/lib64/libQt5Help.so

snowuyl, 2024-11-18:
Thanks for your great support! client, libcore.so, and server can be generated successfully on Rocky Linux 9.4.
$ ls bin
client  libcore.so  libcore.so.0  libcore.so.0.3  libcore.so.0.3.7  server

closed 2024-11-18 by snowuyl