serhmarch/ModbusTools #22: WriteSingleCoil

daBaret, 2025-02-07:
Hi,
thank you for your work and this tool!

When using the "Send message" function I found a small bug with the **WriteSingleCoil** function.
The message is always sent to the coil 1 also if the address is changed.

I believe that the error is here: https://github.com/serhmarch/ModbusTools/blob/e85bb37f2d27bb607fd8907ff8e0ba67de4fe95a/src/client/gui/sendmessage/client_sendmessageui.cpp#L380

If you change the address from `ui->spReadAddress->value()` to `ui->spWriteAddress->value()` then it works as expected.

Have a great day,
Davide

serhmarch, 2025-02-07:
Hello. Thank you for clarifying the issue. I will update it very soon.

serhmarch, 2025-02-07:
[New release](https://github.com/serhmarch/ModbusTools/releases/tag/v0.3.10) is available.

closed 2025-02-10 by daBaret