serhmarch/ModbusTools #25: How to run on Fedora 41 (linux)

VanPhongAutomation, 2025-02-26:
I would like to know how to run this app on Fedora linux. Could you provide a guide?

VanPhongAutomation, 2025-02-27:
Thank you for your quick reply. I did make the Modbus Tools GUI running on Fedora 41.
But the other problems are:
1. How to install converter driver into Fedora (I use USB to RS485 converter).
2. How to determine which serial port are connected to the converter (quite different from Windows OS).
These problems are not related to this thread but if you have solution, please help.

serhmarch, 2025-02-27:
I guess your Linux may recognize the device automatically.

The easiest way to determine which serial port is related to your device is plug it into USB, then open client app and double click on the port node in the project view, choose type of port as RTU and then try to find something like `/dev/ttyUSB0` in the serial port list.

Or you can open client app and watch serial port list, then close app, plug converter into USB, open client app serial port list once again and try to determine which port was added in the serial port list.

VanPhongAutomation, 2025-03-03:
[image]
The serial port connection is denied, is it something that related to Linux "sudo" privileges?

closed 2025-03-03 by serhmarch