serhmarch/ModbusTools #32: MacOS support needed

kelvinji2009, 2025-03-14:
Please add MacOS support.Thanks.

serhmarch, 2025-03-14:
Yep. I guess it will be done someday. But first, I need something with MacOS on board. And, of course, main thing is time.

mattiadevivo, 2025-04-07:
@kelvinji2009 you can use the Docker container on MacOS meanwhile.

Just make sure to have Docker installed, then run in the terminal
```
docker run --rm -p 6080:6080 serhmarch/modbustools:0.4.1
```

and you can connect to the app at `http://localhost:6080/vnc.html`

kelvinji2009, 2025-09-01:
> [@kelvinji2009](https://github.com/kelvinji2009) you can use the Docker container on MacOS meanwhile.
>
> Just make sure to have Docker installed, then run in the terminal
>
> ```
> docker run --rm -p [...]

Good idea and Thanks.