serhmarch/ModbusTools #36: Unable to set an adress past 99

1a11, 2025-03-24:
Hello there! 

Trying to add an address past x99 and unable to do so. Both in the "New item" or "Send message" windows. Input field rejects any letters or digits past 2 symbols. Addresses work fine when force-set in the Address field of an item.  

If you'd need any additional information, let me know

bentstan, 2025-03-24:
same for me.
Using version 0.4.0 let you use addresses past 99. Also if you edit your project in the older version and open it in 0.4.1 it is working, but it shows 99 for every item which has a higher address.

serhmarch, 2025-03-24:
Hello everyone. Thanks for bug report. Already fixed and will be updated in the next v0.4.2 release.

1a11, 2025-03-25:
Acknowledged, will be resolved in 0.4.2

closed 2025-03-25 by 1a11

serhmarch, 2025-04-10:
You can try new [release](https://github.com/serhmarch/ModbusTools/releases/tag/v0.4.2) with the fixed issue.