serhmarch/ModbusTools #38: I use Modbus RTU over TCP a lot. Is it possible to add that feature?

nadeda, 2025-03-28:
I use Modbus RTU over TCP a lot. Is it possible to add that feature?

serhmarch, 2025-03-28:
Hello. Yes. It's possible. I will consider this feature for the next major release.

hagenbuch, 2025-08-26:
Is this the same as Modbus TCP? I am using a Waveshare Modbus RTU / TCP gateway and it works well when using "mbpoll" command line tool.

I'm interested, too!

hagenbuch, 2025-08-26:
Ah OK thank you - also for your quick reply:

Yes, my gateway removes the CRC  apparently and uses the transport layer from TCP. However: How can I scan such a device, say device 1 one IP 192.168.1.1 and port 502? I was able to add a port and then even a device that I know exists.

The only thing I get is this:

    Tx: TCP:192.168.22.12:502,Unit=1: 00 01 00 00 00 06 01 03 1A 34 00 01 
    Rx: TCP:192.168.22.12:502,Unit=1: 00 01 00 00 00 03 01 83 02 
    Info: TCP:192.168.22.12:502,Unit=1 Error (1000002): Returned Modbus-exception with code 2