serhmarch/ModbusTools #41: Add random delay range for server reply

MLaudren, 2025-04-23:
Hi again,

In the Server application,
Would it be possible that instead of a fix delay, we can also select a range, so the answer can vary?

Ex:
Fix time of 20ms,
Or range between 20ms and 100ms, selectable by the user.

Thanks!

MLaudren, 2025-04-25:
That is the part I'm talking about as well
[image]

Can we have the choice of a fix delay (what is the system proposing now), or a range?

So we can tell the application to delay all message of XX ms, or random between XX and YY ms, if the user select a range. 

Thanks

serhmarch, 2025-04-25:
Well, I'll think about it.