serhmarch/ModbusTools #42: MOXA 1100 series fails to read Input registers at Address 1-2

1a11, 2025-04-24:
Trying to do 04x on address:
[image]

Getting:
[image]
Bad Illegal Address

Double checked with Modbuss Poll:
[image]

Device was setup by the Discover feature. Double-checked it, is fine.

1a11, 2025-04-24:
To be perfectly clear, to read ANY data from the device, I had to read a holding register with input register function (03x):

[image]

[image]

serhmarch, 2025-04-24:
Hello. Again. Device address `400001`  is a first `holding register` (or `4x` -register). Read function is number `03`.

If you want to read input register use address like `300001` (`3x` -register, read function is number `04`, as you can see on you own screenshot, Address scan->`Function: 04 Read Input registers (3x)` ).

https://github.com/serhmarch/ModbusTools?tab=readme-ov-file#memory-item-addressing

1a11, 2025-04-24:
Thank you for a quick reply! Thank you, will investigate