serhmarch/ModbusTools #43: Adres number changed bij 1

hooijschuur, 2025-04-25:
I do not exactly know it changed.

But i put in a lot of adres numbers. Checked with modbus mechanic to verify if the responses is at the right adres. After saving . And starting te program the adres is moved bij 1. And i do not know how to change it.

serhmarch, 2025-04-25:
Hello. 

Can you provide more information, e.g. screenshot before and after?

Can I get the file you saved? It is `*.pjc` (for client project) or `*.pjs` (for server project)

serhmarch, 2025-09-24:
Hello @rico666.

> When editing items (takes some time to figure out that by address 0x/1x/... you actually mean FUNCTION CODE)
you choose 0x for function code 1, 1x for function code 2 etc.

No. `0x`, `1x`, `3x` and `4x` it is never a FUNCTION CODE, it is a MEMORY ID. It is an old Modicon notation used from 1980s in its old PLC programming software called `Modsoft` (and then `Concept`).

And yes, it is 1-based!

If you want to switch to 0-based address notations, use menu `Tools/Settings/View/Modbus adr. notation`.

More details about addressing: https://github.com/serhmarch/ModbusTools?tab=readme-ov-file#memory-item-addressing