serhmarch/ModbusTools #52: Is the register order in Register Order and Modbus Poll different in format?

runbrick, 2025-06-23:
Modbus Poll's format "Float AB CD"

In Modbus Tools, the register order is R1R0R3R2 or R3R2R1R0. What's the difference between these two?

What is the correspondence between them?

serhmarch, 2025-06-23:
Hello.

There is no difference for 32-bit data type between R1R0R3R2 and R3R2R1R0, only for 64-bit.

runbrick, 2025-06-23:
I've been using Modbus Poll where the data formats include AB CD, CD AB, BA DC, and DC BA.
What are the different formats corresponding to this software?

serhmarch, 2025-06-23:
This data format can be achived using combination of `Byte Order` (Most or Less Signified First) and `Register order` (for 32-bit use R0R1R2R3 or R3R2R1R0). I'm not using Modbus Poll so I can not tell the exact correspondence.